pololu/minimu-9-ahrs-arduino #8: I'm getting NoN with MinIMU-9 v3 and this library

isaacdrafton, 2015-03-27:
What could be the cause?
Thank you!

isaacdrafton, 2015-03-27:
I managed to make it work with 9600 baud rate. 

I want to use the IMU on a autonomous car for it's heading and I only need the yaw values.
The values remain constant when the imu is standing still but when I'm similating driving in a straight line, the yaw values drift with +/- 10 degrees. I would expect them to remain the same (for the case of a straight line). 
Any idea why?

kevin-pololu, 2015-05-14:
If you have not resolved this, one possible explanation is that the magnetometer is not calibrated well, which would cause inaccuracy in the yaw estimation. I would suggest running the Calibrate example from the LSM303 library again and maybe testing the calibration values with the Heading example. Another possibility is that something in the environment (in the car or its surroundings) has a magnetic field that is affecting the magnetometer's readings.

I'm closing this issue for now because I don't expect this to be a problem with the AHRS program itself. If you continue to have trouble, the [Pololu forum](http://forum.pololu.com/) would be a better place to ask for help.

closed 2015-05-14 by kevin-pololu