pololu/minimu-9-ahrs-arduino #10: OSX compatibility

HamedElhoni, 2015-09-12:
Hi
I'm trying to run this program on MAC. I found Vpython and pyserial libraries but pywin32 not support OSX. what should I do.
Thanks

kevin-pololu, 2015-09-14:
Hi, HamedElhoni.

Unfortunately, the MinIMU-9-test.py program is only intended to run on Windows, and we haven't looked into developing a visualization program for OSX. Note that you can still connect to the Arduino's serial port and see the roll/pitch/yaw angles it outputs as text without the visualization.

mepix, 2017-07-03:
Hi HamedElhoni, Kevin,

This is probably too late to be of any use to you, but I'm going to comment here for anybody else trying to use this tool.

The library **_pywin32 is NOT required to operate this software on Max OS X or Linux!_**  You only need to install pywin32 if you are running it in a Windows environment.  Pywin32 is python for Windows and Mac OS X bundles Python 2.

To get this working on a non-Windows machine:
1. Install the appropriate version of all required packages (except pywin32)
2. Open the file "MinIMU-9-test.py" in the editor of your choice
3. Find the line: "ser = serial.Serial(port='COM9',baudrate=115200, timeout=1)" that selects the serial port used by the software.  You will need to change this line since the "COM9" protocol is for Windows only.
4. For Mac OS X, you can see the available ports by going to the terminal and typing `ls /dev/tty.*` into the terminal.  As a sanity check, you can also go to the Arduino IDE and see what port you used to upload the code to your Arduino.
5. Change the code line so it now reads: "ser=serial.Serial(port='**PATH_TO_YOUR_PORT**',baudrate=115200, timeout=1)".
6. Save the changes to the script and run it.

This should do the trick!

~M

HamedElhoni, 2017-07-03:
Hi mepix
Thank You

closed 2017-07-20 by kevin-pololu

lhcavalcanti, 2018-01-19:
I'm having problem with vpython. After install and update de code to the correct serial port, everytime I run the code, it doesn't find function display.
Any news?