pololu/minimu-9-ahrs-arduino #17: Python program produces no output

moeburn, 2017-01-09:
The Arduino code works GREAT, I'm getting all 3 values in the serial monitor.  But the Python program doesn't seem to work at all.  If you don't have the right COM port in the .py file, it just doesn't load at all, it will display a quick message in the console then close.  So I know that when the program actually stays up, I have entered the right COM port (it also matches the COM port in the Arduino IDE).  I know that my PC is getting the proper values through this COM port because I can see them just fine in the serial monitor.  I've tried both on my PC, and on my laptop, two completely different environments, but in both situations, the Python program doesn't work.  

You get all 3 windows - the compass and heading indicator window, and the 3D representation window, and the console output window, but there's no movement.  The console output window is blank.  The serial log text file it produces is completely blank.  I've tried on two computers, AND with two different versions of Python - I noticed the .py file mentioned 2.6.4 so I went and tried 2.6.4 python and the 2.6 respective versions of all 3 of the addon programs.  On both computers.  

I am absolutely certain that the COM port and baud rate written in the .py file match exactly what they are in Arduino, I am certain that the IMU itself is producing usable values through the Arduino IDE serial monitor, there must be either something wrong with the python code, or there is another addon library/program we are missing, as both of the cases I tried were on clean installs on computers that had never seen Python before.

kevin-pololu, 2017-01-09:
Hi, moeburn.

Python 2.7 should work fine. From the problem you're describing, I think you might be trying to use VPython 6, which causes some issues with the AHRS program, and the easiest workaround is to stick to VPython 5.

For Python 2.7, the direct link to the VPython 5 version you should be using is:
http://vpython.org/contents/download/VPython-Win-Py2.7-5.74.exe

For Python 2.6, you can try:
http://vpython.org/contents/download/VPython-Win-Py2.6-5.72.exe

Let me know if that doesn't solve the problem.

Kevin

AhmadAbdi, 2017-01-15:
Hi there i am facing the same problem as moeburn was, i have followed all the directions and also i have downloaded python from the above directlink which Sir Kevin said one should be using. but problem is still there ??

kevin-pololu, 2017-02-04:
I'm going to close this issue for now as I don't think there is an problem here with the program code, just a matter of getting the right dependencies installed.

AhmadAbdi: as I mentioned your other issue, the [Pololu forum](https://forum.pololu.com/) is a better place for troubleshooting (or you can [contact us directly](https://www.pololu.com/contact)). If you are still having trouble, it would be helpful if you could tell us the exact versions of the software and libraries you installed so we can verify you have the right ones.

Kevin

closed 2017-02-04 by kevin-pololu