pololu/minimu-9-ahrs-arduino #19: No Output on Arduino Serial Monitor

RailsKO, 2017-02-02:
I have been working a bit to set up the MinIMU-9 v3 Gyro, Accelerometer, and Compass (L3GD20H and LSM303D Carrier) with Arduino and Python. After some time I now have the correct versions on Python and Arduino so that the Python software comes up, yet does not show any points of freedom. Neither does the serial monitor.
The serial monitor only displays Pololu MinIMU-9 + Arduino AHRS

I am at the correct port, (COM4) and baud, (115200) with the connections
 Arduino   IMU
   -------   ---
    3V3 - VIN
    GND - GND
    SDA - SDA
    SCL - SCL

Any Help ?

RailsKO, 2017-02-03:
?

kevin-pololu, 2017-02-04:
Hi, RailsKO.

Please post on the [Pololu forum](https://forum.pololu.com/) or [contact us directly](https://www.pololu.com/contact) so that we can help you troubleshoot.

GitHub issues are for tracking bugs or improvements concerning the software in the repository; what you're seeing doesn't sound like a software bug at this point, so I'm going to close the issue for now.

closed 2017-02-04 by kevin-pololu