pololu/minimu-9-ahrs-arduino #21: Arduino MEGA 2560 with AltIMU-10 v5

sundayboyz, 2017-09-06:
Hi, I'm currently using Arduino MEGA 2560 with AltIMU-10 v5 but it is currently nt showing me any output values in Arduino Serial Monitor. Any idea why?

sundayboyz, 2017-09-06:
I've found a solution!

Delete away everything regarding L3G & LSM303 and #define IMU_V5

Hope this helps!

muhammadumerkhan7, 2017-09-11:
i m using miniIMU 9 v5 with arduino UNO and it shows garbage values in the output ... can  u help please ?? kindly send me your modified code

sundayboyz, 2017-09-15:
Hi, here r the modified codes.

https://forum.arduino.cc/index.php?topic=500294.0

Rmb to calibrate your miniIMU.

muhammadumerkhan7, 2017-09-24:
it still isnt working shows garbage values as output ///..... i ve tried
simple magneto meter library and accelero meter library to check if my
sensor is working and it works fine

muhammadumerkhan7, 2017-09-24:
[image]

kevin-pololu, 2017-10-21:
@muhammadumerkhan7 The garbage output you are seeing suggests that you have the wrong baud rate selected in the Arduino serial monitor. The AHRS program uses a baud rate of 115200; unfortunately, I can't tell from your screenshot if you have the correct value selected. If you are still having trouble, please post a question on the [Pololu forum](https://forum.pololu.com/).

@sundayboyz It should not be necessary to delete anything as long as you uncomment `#define IMU_V5`; the sketch should only look for the libraries needed for the IMU version that you specify.

closed 2017-10-21 by kevin-pololu