pololu/minimu-9-ahrs-arduino #26: Windows freeze

Riteshlogin, 2020-06-12:
I'm using a Teensy 4.0 connected to the MiniIMU-9 v5, and can get printouts from it just fine. However, I've tried to use the python application to get an idea of how good this module is at measuring orientation, and the windows the python module generates keep freezing, and if I click on them eventually "Not Responding" shows up in the title bar. I have uncommented the #define IMU_V5, and because I was afraid the Teensy 4.0 might be too fast, I added a one millisecond delay after the print statements. In the terminal, the values I'm getting seem to make sense (as in I can independently change r,p, and y by moving the module around) and I never get "Invalid line".

kevin-pololu, 2020-10-24:
If you are still looking for help getting this to work, could you check that you have the required versions of everything installed? Unfortunately, this Python program is now quite old and has not really been updated, so you need some very specific (and usually older) versions of all the software for it to run correctly. I was able to get it working today with:
- [Python 2.7.18 (32-bit)](https://www.python.org/ftp/python/2.7.18/python-2.7.18.msi)
- [pywin32 228](https://github.com/mhammond/pywin32/releases/download/b228/pywin32-228.win32-py2.7.exe)
- [pyserial 2.7](https://sourceforge.net/projects/pyserial/files/pyserial/2.7/pyserial-2.7.win32.exe/download)
- [VPython 5.74](http://vpython.org/contents/download/VPython-Win-Py2.7-5.74.exe)

These links should go directly to the installer downloads.

Riteshlogin, 2020-10-26:
Ah, that's probably the root of the issue. Thank you for that info, I'll try that out.